UltimateHackingKeyboard/uhk60v1-electronics #10: Re-wrap and reposition the "Unscrew the 5 large below the keycaps..."

spuder, 2015-12-10:
Re-wrap and reposition the "Unscrew the 5 large below the keycaps and the 1 screw on the PCB" text to be more readable

spuder, 2015-12-10:
@mondalaci Could I get some feedback on what you are looking for here? 

Here is one proposal that rephrases for less characters and moves text a little further away from the vias / holes. 

[image]
[image]

mondalaci, 2015-12-10:
@spuder Your suggested solution looks great! I'd love to merge a pull request implementing the above.

closed 2016-01-05 by spuder