UltimateHackingKeyboard/uhk60v1-electronics #15: Cherry MX LED lights

spuder, 2015-12-10:
Allow modders to use cherry mx switches with LEDs

- [ ] Update library to use cherry mx pinout with led holes. [example](http://cherrycorp.com/wp-content/uploads/2014/12/11.gif)
- [ ] Can USB power 100+ LEDs? May need to add an led driver and/or use PWM to reduce power requirements

mondalaci, 2016-01-05:
It'd be useful to take a look at some open source keyboard designs that have backlighting. I'd rather go for a simple implementation that supports a single PWM channel for all the LEDs. So I mean no per-LED backlighting.

spuder, 2016-01-18:
Some possible led drivers

[74HC595 Shift Register](https://www.adafruit.com/products/450) 
- only needs 3 pins (SPI)
- can daisy chain
- Have to do charlie plexing in software

[MAX7219 Led Driver](https://www.adafruit.com/products/453)
- Handles control and refreshing. Greatly simplifies code
- Kind of expensive.

I'm leaning towards the first option because it is cheaper. 

The teensy has an spi library https://www.pjrc.com/teensy/td_libs_SPI.html

mondalaci, 2016-01-19:
I'm aware of these options. Unfortunately, the 74HC595 only allows for a single LED per pin and the MAX7219 is fabulously expensive.

I've been suggested the TPS61193 lately which may work for us. We could string about a dozen of LEDs per OUTx pin.

[image]

I personally don't feel confident in my abilities to choose the best part for this purpose and will surely talk to an electronics engineer before making a choice. But suggestions are always welcome!

spuder, 2016-01-19:
Looking at the [ergodox infinity](https://github.com/kiibohd/pcb/tree/master/ICED%20-%20Left), they are using the  [IS31FL3731C](http://www.mouser.com/ProductDetail/ISSI/IS31FL3731C-SALS2/?qs=rnuWwpjhhVvfjmOe6D9bPQ%3D%3D)

mondalaci, 2016-01-19:
Yes, I know, but [nobody has it on stock](https://octopart.com/search?q=IS31FL3731C) which is not very promising.

mondalaci, 2016-02-09:
Apparently, this IC is [actually available](https://octopart.com/search?q=IS31FL3731) without the closing C in its part name.

mondalaci, 2016-09-27:
Done. We went with the IS31FL3731-SALS2.

closed 2016-09-27 by mondalaci