UltimateHackingKeyboard/uhk60v1-electronics #3: Replace 6P4C connectors with 4P4C

spuder, 2015-12-10:
Replace 6P4C connectors with 4P4C. Make sure that the chosen connector and the wall of the case leaves enough space for the clip so that it can be pressed. Also, 4P4C probably almost always uses reverse wiring!

mondalaci, 2015-12-13:
@spuder Thanks for confirming reverse wiring! I meant to use reverse wiring for 4P4C instead of the current straight wiring because according to my experience telephone cord cables use reverse wiring.

I'd love to back up the prevalence of reverse wiring with facts beyond my (very limited experience) but couldn't find any stats of the percentage of telephone cords with straight vs reverse wiring. The type of wiring shouldn't matter for telephones but it does matter for the UHK.

Andras always does mechanical interference checks in CAD so that shouldn't be an issue.

spuder, 2015-12-13:
Sorry, I can't confirm the reverse wiring as I don't know anything about that part. The comment was copied and pasted directly from the TODO

>-* Replace 6P4C connectors with 4P4C. Make sure that the chosen connector and the wall of the case leaves enough space for the clip so that it can be pressed. Also, 4P4C probably almost always uses reverse wiring!

https://github.com/UltimateHackingKeyboard/electronics/commit/c4327e8296f0e911c573165dafe6d01c02b7f867#diff-b7b1e314614cf326c6e2b6eba1540682L2

mondalaci, 2015-12-13:
So I commented my own comment - that's hilarious! :)

Sorry, it's my bad, and will look into it another time.

mondalaci, 2016-09-27:
Done.

closed 2016-09-27 by mondalaci