UltimateHackingKeyboard/uhk60v1-electronics #6: Reduce resistor values on the TX and RX lines

spuder, 2015-12-10:
Reduce resistor values on the TX and RX lines of the board-to-board connectors because they didn't work on the 5.1 generation prototypes

spuder, 2015-12-13:
R7 => 10K
R8 => 10K
R9 => 100K
R10 => 10K

What values should these be?

mondalaci, 2015-12-13:
Great question! I'm fairly clueless about analog electronics so a professional electronics engineer will revise the design before manufacturing and fix such issues.

In any case, I should add detailed instructions about using the UGL and Opendous KiCad libraries because they're not added by default, which can be seen on the attached picture.

spuder, 2015-12-13:
I was just asking about those libraries in #22

mondalaci, 2016-09-27:
This has been redesigned.

closed 2016-09-27 by mondalaci