UltimateHackingKeyboard/uhk60v1-electronics #24: Create automated system to compare kicad diffs

spuder, 2015-12-14:
When a new merge request is made. Use travis-ci to generate gerbers/pdf/png files and then use `linux compare` to generate a visual diff. 

http://www.evilmadscientist.com/2011/improving-open-source-hardware-visual-diffs/
https://github.com/blairbonnett-mirrors/kicad/blob/master/demos/python_scripts_examples/plot_board.py

- [ ] Create travis config file
- [ ] Create scripts to generate diffs based off of plot_board.py
- [ ] Create aws account to upload artifacts to s3

You can assign this to me as I'm currently working on it.

mondalaci, 2015-12-14:
Awesome thought of automating this!

Just made you a collaborator of this repo with write access, and assigned this issue to you.
Please let's not change the board files until the visual diff mechanism is in place. I'm very afraid of the board changes as I can't see them and they may result in broken boards which translate to huge problems / losses upon manufacturing the PCB.

Sorry for the delay regarding the other issues, I'll catch up later on.

Thank you so much!

spuder, 2015-12-14:
Awesome thanks. 
I understand that you have plenty of other things to worry about right now. 
I have some manual diffs ready now, but I'm ok waiting to merge them until we have a more polished diff system in place.

spuder, 2015-12-16:
I have a working implementation in this branch in my fork

https://github.com/spuder/electronics/tree/feature/wip_travis

It still has the following issues: 

- Granting public access to the s3
- Minor updates to the diffs
- Decide if travis should generate gerbers too

timofonic, 2015-12-19:
Are there a possibility of making this KiCad native? It would be a great addition for KiCad users!

It would be great if someone implements Git support into KiCad, then auto-generate visual-diffs...

I would like to "hack" your schematics. Your project seems interesting and I would like to donate something, but I'm unable to spend so much money so maybe I can contribute in possible hardware updates or forks :dancer:

spuder, 2015-12-19:
Stand by, i'll be pushing a script soon that uses vagrant and the kicad python api to make images showing diffs. Its as native as can be right now

spuder, 2015-12-21:
I have opened merge request #27 It has a fairly polished collection of scripts that when run generate 14 pdf files, converts them to .png files, and then runs `compare` to show the changes

    ./scripts/generate-png-diff.sh

The script that generates the plots is located externally here: 

https://gist.github.com/spuder/4a76e42f058ef7b467d9

It still could use some minor improvements, but it is good enough to see at a glance exactly what was changed. 

It also has a travis config file, that runs these scripts. 

Here is what happens if nothing has changed: 

https://travis-ci.org/spuder/electronics/builds/98057604

Here is what it looks like when a change is made to a file ending in .kicad_pcb

https://travis-ci.org/spuder/electronics/builds/97929762

There are still two remaining tasks

- Automatically upload the .png artifacts to amazon s3 or imgur.com on commit
- Have travis-ci post a link to the generated files inside the comments section of github. 

I'll keep working on those, but I want to merge what I have now so that people can use these scripts without waiting. 

Here are some samples of what happens when the kicad_file is edited. Here I intentionally moved the rj11 connector: 

[image]

[image]

mondalaci, 2015-12-21:
I'm absolutely thrilled about this, thank you so much @spuder! Let me merge the relevant pull request real quick.

mondalaci, 2015-12-21:
@timofonic Sure thing, everybody's welcome to hack the schematic! Let me know if you need any help.

spuder, 2015-12-23:
Marking as done.
Having travis upload the .png files to imgur and automatically posting links in the github comments wont have much use, since travis secrets are not persisted across forks. I won't automate that unless I there is a real need, and I figure out a way that works with forks. 

I'll update the README instructing users to upload their diffs manually.

closed 2015-12-23 by spuder

mondalaci, 2015-12-27:
@spuder Thank you for your hard work, this looks excellent the way it is! The rest is only icing on the cake and certainly not urgent.

hoijui, 2020-09-09:
As an alternative, you might want to have a look at the very powerful [KiBot](https://github.com/INTI-CMNB/KiBot) tool.