UltimateHackingKeyboard/uhk60v1-electronics #30: Want to add the opensource logo to the board?

spuder, 2016-01-20:
Thoughts on adding the official open source logo to the silkscreen of pcb? 

http://www.oshwa.org/definition/

[image]

https://www.youtube.com/watch?v=QrHoowPpdQ8

mondalaci, 2016-01-20:
Correct me if I'm wrong but I think the UHK is not fully open according to OSH principles because we haven't yet released the CAD files.

spuder, 2016-01-20:
Good point, didn't think of that. 

I've read it a few times, and I see how it could be argued both ways. As a whole, the keyboard doesn't meet the criteria yet. The pcb most likely does. I don't know if the enclosure counts.

>The program must include source code, and must allow distribution in source code as well as compiled form. Where some form of a product is not distributed with source code, there must be a well-publicized means of obtaining the source code for no more than a reasonable reproduction cost preferably, downloading via the Internet without charge.

I'll check with the mailing list to get clarification on this principle. I don't think I'll pursue this much further.

closed 2016-01-20 by spuder