UltimateHackingKeyboard/uhk60v1-electronics #38: Travis script throwing error

zen, 2018-12-27:
I'm trying to use the travis diff script and it's throwing error:
root@12ba8a8b9dbe:/kicad# ./generate-png-diff.sh
Copying 187bd8e:drone-kicad-sample.kicad_pcb to ./plot/187bd8e/
Kicad files saved to:  './plot/current' and './plot/187bd8e'
Converting ./plot/current/drone-kicad-sample.kicad_pcb to .pdf:  Files will be saved to /tmp/pdf
Traceback (most recent call last):
  File "/tmp/plot_board.py", line 172, in <module>
    popt.SetColor(YELLOW)
NameError: name 'YELLOW' is not defined
Converting ./plot/187bd8e/drone-kicad-sample.kicad_pcb to .pdf's Files will be saved to /tmp/pdf
Traceback (most recent call last):
  File "/tmp/plot_board.py", line 172, in <module>
    popt.SetColor(YELLOW)
NameError: name 'YELLOW' is not defined

zen, 2018-12-27:
@spuder probably something for you to take a look.

spuder, 2018-12-27:
Most likely what is going on is an incompatible version of a package that is installed. The travis scripts are only tested on ubuntu 14.04 (trusty).

Are you running these scripts on your workstation? Or is it travis that is failing? 

Do you have docker installed? Could you try running these commands and see if it gives the same error? 

```
docker run -it --rm -v $(pwd):/data ubuntu:xenial /bin/bash
apt-get update
apt-get install -y sudo software-properties-common
cd /data
./scripts/1-add-repos.sh
./scripts/2-install-dependencies.sh
./scripts/generate-png-diff.sh
```

zen, 2018-12-28:
Hi, thanks.
With manual steps it's even worse:
`root@f9e6f20d857e:/data# ./scripts/generate-png-diff.sh
Copying 187bd8e:drone-kicad-sample.kicad_pcb to ./plot/187bd8e/
Kicad files saved to:  './plot/current' and './plot/187bd8e'
Converting ./plot/current/drone-kicad-sample.kicad_pcb to .pdf:  Files will be saved to /tmp/pdf
14:10:10: Debug: Skipping general section token 'drawings'
14:10:10: Debug: Skipping general section token 'tracks'
14:10:10: Debug: Skipping general section token 'zones'
14:10:10: Debug: Skipping general section token 'symbol'
Traceback (most recent call last):
  File "/tmp/plot_board.py", line 58, in <module>
    board = LoadBoard(filename)
  File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 12870, in LoadBoard
    return _pcbnew.LoadBoard(*args)
IOError: KiCad was unable to open this file, as it was created with a more recent version than the one you are running. To open it, you'll need to upgrade KiCad to a more recent version.

Date of KiCad version required (or newer): 11/30/17

Full error text:
KiCad was unable to open this file, as it was created with a more recent version than the one you are running. To open it, you'll need to upgrade KiCad to a more recent version.

Date of KiCad version required (or newer): 11/30/17

Full error text:
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source
'./plot/current/drone-kicad-sample.kicad_pcb'
line 172
offset 16
from dsnlexer.cpp : Expecting() : line 369

Converting ./plot/187bd8e/drone-kicad-sample.kicad_pcb to .pdf's Files will be saved to /tmp/pdf
14:10:10: Debug: Skipping general section token 'drawings'
14:10:10: Debug: Skipping general section token 'tracks'
14:10:10: Debug: Skipping general section token 'zones'
14:10:10: Debug: Skipping general section token 'symbol'
Traceback (most recent call last):
  File "/tmp/plot_board.py", line 58, in <module>
    board = LoadBoard(filename)
  File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 12870, in LoadBoard
    return _pcbnew.LoadBoard(*args)
IOError: KiCad was unable to open this file, as it was created with a more recent version than the one you are running. To open it, you'll need to upgrade KiCad to a more recent version.

Date of KiCad version required (or newer): 11/30/17

Full error text:
KiCad was unable to open this file, as it was created with a more recent version than the one you are running. To open it, you'll need to upgrade KiCad to a more recent version.

Date of KiCad version required (or newer): 11/30/17

Full error text:
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source
'./plot/187bd8e/drone-kicad-sample.kicad_pcb'
line 171
offset 16
from dsnlexer.cpp : Expecting() : line 369

Converting /tmp/pdf/current/*.pdf to .png
I/O Error: Couldn't open file '/tmp/pdf/current/*.pdf': No such file or directory.
Converting /tmp/pdf/187bd8e/*.pdf to .png
I/O Error: Couldn't open file '/tmp/pdf/187bd8e/*.pdf': No such file or directory.
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-AssyOutlinesTop-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-AssyTop-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-CuBottom-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-CuTop-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-EdgeCuts-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-Layout-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-MaskBottom-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-MaskTop-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-PasteBottom-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-PasteTop-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-Silk-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-SilkBottom-1.png
Generating composite image ./plot/diff-current-187bd8e/drone-kicad-sample-SilkTop-1.png`

I cannot use 14.04 due to very old KiCAD package. I checked KiCAD source and cannot find where YELLOW and other colors are defined. It's very strange that other literals work and only YELLOW is missing
I think i might just change it to some other color.

By the way, are you going to maintain this script? Maybe it's worth separate repo?

spuder, 2019-01-04:
Still looking into this. The original purpose was to allow the UHK developers to visually see any changes made to the design in merge requests. It needs to be part of this repo so Travis-CI could run the script. 

The root cause is that the designs have moved to kicad5, whereas the scripts were written for kicad4.

mondalaci, 2019-01-04:
We don't need the visual diff feature anymore, so feel free to remove it @spuder. Thanks for your help!

closed 2019-06-20 by mondalaci